WPIRoboticsEngineering/ZenRobotBuildKit #17: Instructions for shim board

gcl8a, 2020-08-07:
Obviously, it's on your plate, so this is a placeholder, but also questions. My plan is to solder the power board with:

- (2) 6-hole sockets for the motors
- (2) 6-pin headers in the "other" motor holes
- (1) 6-pin header for the pwm/dir/slp
- (1) 3-pin header for power

Anything else?

Then for the shim board:
- (2) 1x20(?) sockets to hold the DEV board
- (2) 1x20(?) sockets for the I/O
- (1) 1x4 pin header for the servo
- (1) 1x4 socket for the ultrasonic

Not sure about:
- line sensors
- disable motor
- there appear to be 7 spots for diodes, but I only have 6. are they all used?

Then solder the shim to the power board. Note that we have to get the shim correct before it gets soldered in, as they'll never be able to desolder that if needed. Pins and sockets would be better so they can remove the shim, if needed. Not sure if that fits with the motors.

madhephaestus, 2020-08-07:
ist 1x19 for the ESP, btw

disable motors is a 3x1 male, we provide a jumper in the kit

oops on the num of diodes, you can safely ignore the diode nearest the ESP32, thats the clamp for the servo voltage feedback, and it it redundant at this point since that sensor doesnt go high enough to be a problem

there is an unlabeled resistor also near the servo header, thats the in-line resistor for the ultrasonic

As for solder order, i reccomend soldering up the red board except all the interface pins, and then solder the interface pins while they are all loose so there is no allignment issue during assembly.

gcl8a, 2020-08-07:
What size resistor did you spec? I don’t think I got one.

Are these diodes 3.3V or 3.9V? If the latter, I’d say we use the same sized resistor as the pullups on the encoders, yes?

From my understanding of zeners, the currents are way too small to get them up near their nominal voltages. Typically, you shoot for several mA minimum, and we’re at 500 uA with a 10k pullup resistor. That would explain why a 3.9V zener is still only at ~3V.

madhephaestus, 2020-08-07:
i used a 10k

madhephaestus, 2020-08-07:
arg, we will need to swap the encoder lines for one of the motors in the header file to use the shim.

gcl8a, 2020-08-07:
No big deal. It’ll be a whole new term.

madhephaestus, 2020-08-07:
tested:
both motors and encoders
ultrasonic
servo
web control

On the SHIM board. the only error so far is the swap of the encoder pins. 

Before soldering the battery tabs, the system can be tested while the solder connections are easy to get to. After the system is tested, then the driver can be soldered to the battery terminals.

gcl8a, 2020-08-09:
On mine, the ECHO pin seems to be connected directly to pin 22 on the DEV module header, in addition to pin 17 via the resistor.

Minor note: all of the diode pads are “reversed” in that the positive side is square, which is normally reserved for GND.

gcl8a, 2020-08-26:
My order for soldering -- which went well -- is:
- Solder all the headers and pins and diodes on the shim board EXCEPT the ones that connect it to the power board
- Solder the 1x6 motor connectors on the power board
- Put the pins in the power board and place the shim on top (for alignment)
- Solder the pins in the power board
- Remove the shim and solder the power board to the chassis (battery tabs)
- Place the shim board back on the power board and solder the pins
- You may need to slightly bend some pins so they don't interfere with the motors. Touching is OK, but they might prevent the motors from fully seating.

gcl8a, 2020-08-29:
@madhephaestus Students are getting their kits. We need final instructions for assembly.